aotta/PiRTOII #1: Recommended edge connector option?

hadess, 2024-06-26:
I guess this question stands for most of your "flashcarts": which PCB thickness and edge connector finish do you recommend using? The defaults for my supplier are 1.6mm thick PCBs, and ENIG edge connector finish. Are those OK? What bevelling would you recommend?

aotta, 2024-06-26:
I used defaults also, in my case 1.6mm thick but Hasl with lead finish.
Enig should works fine too, but it's usually more expensive.
Gold fingers also may be a good options, but i wanted to keep all my
flashcart cheaper as possible.

hadess, 2024-06-26:
The ENIG finish was relatively cheap (compared to shipping or VAT) and I went with 30 degrees bevelled edge as recommended here: https://www.retrorgb.com/determining-the-build-quality-of-a-game-cartridge.html

Thanks!

closed 2025-03-10 by aotta