aotta/PiRTOII #3: A couple of design questions

SukkoPera, 2024-06-30:
I'm working on a modified version of this cart that used the original Pico board (just because I have plenty of those) and I have a few questions on the design. Please note that I don't know anything about the Intellivision. I have bought one some time ago but I have no carts, hence my interest in this project :). So please excuse me if my questions are naive, it's really the first time I look at this stuff:

* I see that pins 8 and 10 are permanently grounds. I can understand it for pin 8, which, according to [the pinout](https://wiki.intellivision.us/index.php/Cartridge_Port) is the EXT_VIDEO input (I guess it's for reducing noise), but not for pin 10: it looks like it is the MCLK _output_, and forcing it to ground doesn't sound good to me. I think it should just be left unconnected. Maybe you wanted to ground pins 6 (EXT_AUDIO) and 8 but you miscounted? Just a thought :).
* The reset circuit does not look correct to me. The BC547 is an NPN transistor and seeing its collector connected to GND is very odd. Can you please explain the rationale behind it?

Thanks a lot! :)

SukkoPera, 2024-06-30:
This would be my proposal for the reset circuit:
[image]

(_reset_inty_out_ is the signal coming from the Pico, I've just renamed it)

SukkoPera, 2024-12-22:
I have finally tested my above proposal for the reset circuit and I can now say that it works perfectly! ;)

aotta, 2024-12-22:
Thank @SukkoPera for the alternative design, so close the issue

closed 2024-12-22 by aotta