aotta/PiRTOII #4: Black screen on power up

repetto74, 2024-07-05:
Hi,

I have built this project. All soldering is good and checked. I used a diode 1N4148 and a BC547 transistor. No pushbutton installed for the reset.
Pico programmed from USB and bin files copied in the "PICOII" drive showing up on the desktop after firmware programming.

I get a black screen with a background video noise when I power up the Intellivision. This is the normal behaviour when no cart is inserted in the console.
Pico led is not lighting up. Looks like the Pico does not get power. Any idea what this could be?

aotta, 2024-07-05:
one user experimented similar issue due to not recognizing MSYNC signal at
boot, you could try the alternate firmware i just uploaded if it help

repetto74, 2024-07-05:
Ok thanks I will give a try with the alternate firmware thanks :-)

repetto74, 2024-07-05:
I reprogrammed the pico with the alternate firmware (noticed that games binaries copied in the Pico are not cancelled but I assume this is normal).
Unfortunately same behaviour. Black screen with video noise in the background.

repetto74, 2024-07-05:
The pico is the purple clone as stated by the Ali merchant. Still a doubt maybe on the pinout of this one but I think it is the good one.
This is the one I picked for this project: https://www.aliexpress.com/item/1005007052696317.html?spm=a2g0o.order_list.order_list_main.5.3cc318022FPEpg

aotta, 2024-07-05:
So, sorry but that Pico clone has a different pinout than the usual
"purple" clone and it doesn't work with PiRTO.
As you could see, for example it has GND at pin no. 3 while the right one
has GND at pin no. 6, as better showed in the repository from which i was
inspired: https://github.com/robinhedwards/A8PicoCart

repetto74, 2024-07-05:
Ok thanks now it is clear :-) ! I need to source the good Pico clone.

repetto74, 2024-07-05:
I finally found the good purple pico and ordered. This should fix the issue :-).

https://www.aliexpress.com/item/1005004277013272.html?spm=a2g0o.order_list.order_list_main.5.73e41802q3bytF

emilen72, 2024-07-06:
Hi, I have the same behavior and I built the cart with the same latest purple clone you ordered. The cart works but I need every time:
1) insert the cart
2) turn on the console
3) reset the console
4) reset the cart (install button or short with tweezers)

then the start menu and game are listed.
Remember to add the 0.cfg file in the root otherwise the games won’t start.

hope this help

aotta, 2024-07-06:
The user at atariage solved with the alternative FW in published yesterday, please @emilen72 check it you too

aotta, 2024-07-06:
A Pico clone with different pinout will never work without changes in PCB, and also trying it could damage your Pico and/or the console!

mamejay, 2024-07-07:
> The user at atariage solved with the alternative FW in published yesterday, please @emilen72 check it you too

PERFECT!! This works well.

emilen72, 2024-07-08:
> The user at atariage solved with the alternative FW in published yesterday, please @emilen72 check it you too

Ok, it work… only noted that on every game, after that the game start well, the led blink 1 time endless… its ok too?

bataais, 2025-01-31:
Unfortunately USCF Chess 'hangs' at white move 21 on the PiRTOii. Clocks are counting, but no move is made.
I have the right .bin and .cfg files.
 
Try it with C vs C, Color (Left Controller ENT, ENT, 8).
 
Seems a RAM problem..

closed 2025-03-10 by aotta