aotta/PiRTOII #6: Is core1 necessary ?

dombru54, 2024-08-07:
You use Core1, but why ? 
Code : multicore_launch_core1(core1_main);
Is Core0 more sensitive to interrupts or others hardware interactions ?
Thanks for your expertise.

aotta, 2024-08-07:
Both core are needed, on core1 i run the bus emulation, that had a timing
aligned with console and can't stop never. On core 0 run all other stuff.

dombru54, 2024-08-07:
I understand now the uncommented line  :
multicore_launch_core1(core1_main);	
Maybe, i can add comment :	// Core1 run the unique function "core1_main()"

then, in function "void __not_in_flash_func(core1_main())", you must specify this function :
multicore_lockout_victim_init();
Maybe, i can add comment :	// Core1 is obliged to do a pause in knew state by Core0.

Is Core0 more sensitive to interrupts or others hardware interactions than Core1 ?

closed 2025-02-22 by aotta