aotta/PiRTOII #9: Question about rPico pinouts

8lall0, 2024-10-27:
I would like to ask what are the differences between the original rPico and the purple clone :)

Thanks!

crystalct, 2024-11-06:
Could you tell us where to read the characteristics of this clone which make it safe for the use of 5V in the GPIOs.....

SukkoPera, 2024-12-22:
It is never clearly stated, but I think that the board was designed for the purple clone because it can be bought with much larger flash size than the official board (16 MB vs 2 MB), thus more ROMs can be put on it.

The clone has no special characteristics that make it 5V-tolerant, it's just the RP2040 that [is *unofficially* 5V-tolerant](https://hackaday.com/2022/09/25/pi-pico-w-does-pcmcia-gets-this-ibm-pc110-online/#comment-6516402).

Anyway, you might be happy to know that I have designed [a variant of this board](https://github.com/SukkoPera/PiRTOII) that uses the official Pico and (re)introduces proper level shifting. Feel free to check it out! ;)

aotta, 2024-12-22:
I confirm the only real reasons why i used the purple clone is the 15mb flash free instead of the 780~kb of the green one.
Thank to @SukkoPera for his variant, now anyone could choose his favourite flavours.

closed 2024-12-22 by aotta