aotta/PiRTOII #10: Games not loading, but Booting OK to the menu screen.

roninfourseven, 2024-10-31:
I have built a PiRTOII using PCBWay to manufacture the PCBs from the gerbers on here.

The Pi is a purple Pi from Ali Express and I bought two, one is installed in one of your PiColecos and works fine BTW.

The PiRTOII boots fine to the menu screen and I can navigate to ROMs (These have a BIN extension and the filenames are less than 32 characters), however whin I select a file it sits with the green light pulsing and the Intellivision is frozen.

The Intellivision works fine with normal carts.

I have tried both firmware versions but get the same and I have reflowed all the soldering.

One thing, my Intellivision is a UK model and I don't know if the ROMs are region sensitive.

Any help would be useful, thanks

Jon

roninfourseven, 2024-10-31:
I didn't, can you elaborate a little more please

MMarcoux66, 2024-10-31:
Look at this thread...  https://forums.atariage.com/topic/366700-pirto-ii-multicart-just-published/page/2/

roninfourseven, 2024-10-31:
OK so 0.cfg has been added and it boots, but ends up at a black screen with the Purple Pi flashing it's light every other second.

roninfourseven, 2024-10-31:
Good news, I downloaded the cfg files from atariage and it seems to be working :)

aotta, 2024-10-31:
Happy you solved, closing.

closed 2024-10-31 by aotta