aotta/PiRTOII #14: Richiesta informazioni su progetto PIRTOII

Fing972, 2025-02-21:
Salve,
innanzi tutto complimenti per il progetto, avevo già iniziato a seguire la versione 1 ma, ora che mi sono deciso a volerla realizzare, vorrei direttamente  puntare alla versione 2 ma avrei necessità di alcuni chiarimenti, se possibile, dato che non sono particolarmente esperto.
Per prima cosa ho studiato il PCB e ho visto che è realizzato con un pico purple che dovrebbe supportare i 5v.
Ho letto che ufficialmente il pico lavora a 3.3v, io ho acquistato in passato dei pico e sono la versione yd-rp2040 a 16mb che probabilmente hanno un pinout da confrontare bene con il purple e forse dovrei inserire dei level shifter 3.3-5v per essere sicuro.
Il sorgente della prima versione del PIRTO era basato sull'arduino IDE (.ino) mentre la seconda versione ho dovuto capire come funziona l'SDK per raspberry, poco male, un po di esperienza in più e dopo un po sono riuscito a compilare il tutto.
Qui arrivano i dubbi.
Le rom nella prima versione venivano messe nella sdcard mentre non ho capito bene nella seconda come vengono gestite e come viene utilizzato il pacchetto intybasic con dentro i sorgenti basic per i title.
Grazie in anticipo per i chiarimenti.
Saluti.
Alessandro

aotta, 2025-02-22:
Una volta caricato il firmware, se connetti la pirto al pc via usb (non inserita nella console) si apre una finestra con la cartella flash e puoi copiare i file bin+cfg con un semplice drag & drop. 
Il codice intybasic è per il menu ma non ti serve modificarlo

closed 2025-02-22 by aotta

Fing972, 2025-03-05:
Salve,
ho realizzato un prototipo utilizzando però un pico yd-rp2040 con 16mb.
ho adattato la piedinatura e modificate le impostazioni per il tipo di memoria.
La compilazione è andata a buon fine, ho scritto il firmware e sull'intellivision appare la schermata principale.
Il problema che ho è che una volta effettuato il flash del firmware il pico non mi viene visto, una volta collegato al PC, come memoria di massa e quindi non riesco a copiare le rom. Se vado in bootsel e copio qualcosa, naturalmente poi i dati vengono persi.
Il pc ad un certo punto mi da un errore tipo usb non riconosciuto.
Se invece metto il firmware del micropyton, dopo il flash me lo riconosce come memoria di massa di circa 15mb.
Cosa potrebbe essere andato storto, ho provato tutte le versioni di firmware.
Forse qualche impostazione del flash_fs ?
Grazie.

aotta, 2025-03-05:
Il sistema "doppio" è gestito con le librerie TinyUSB, devi installarle e selezionarla come USBStrack nel compilatore Arduino IDE